18r441m/CSCE274 #8: hw 2 - Can't find release file

angelaalaisa, 2021-09-08:
When I tried to run` sudo apt-get update` it gave me a not found error for the release file in the repository for 'https://download.docker.com/linux/ubuntu \ Release". Everything seemed to be working fine up until this point. How do I get the missing file?

18r441m, 2021-09-09:
Did you ensure you an internet connection. Also, you can paste the error here or post screenshots.

angelaalaisa, 2021-09-09:
[image]
 Yes, I was connected to the internet and had no problems downloading anything before running that command.

ZeHolyQofPower, 2021-09-10:
I don't know if you succeeded in installing docker and it's dependencies, but I know Ubuntu has a graphical way of installing a lot of common apps using it's "Software" app. You could check there. If not, there are often multiple ways to install things on Linux, some very hard, some kinda easy.

angelaalaisa, 2021-09-10:
I was able to install it, I think I just mistyped something. But I'll definitely look into the app just in case I run into problems again in the future, thanks for the suggestion!

closed 2021-09-10 by angelaalaisa