18r441m/CSCE274 #12: Bridging over ethernet to the Duckiebot

marksoehner, 2021-09-15:
We are following the steps in the documentation but when we try to ssh to the duckiebot, the terminal shows an error message.  The error message is "could not resolve the hostname".

18r441m, 2021-09-16:
could you attach the exact command you ran. Also, did you use <dts fleet discover> to identify the duckiebot?

MMAzeem3, 2021-09-16:
I's getting the same error. We booted the duckiebot and saw that is was ready using 'dts fleet discover'. When I try to ssh into it using 'ssh duckie@duck28.local' I get 'ssh: Could not resolve hostname duck28.local: Name or service not known'. I also can't access http://duck28.local/ or http://duck28/. I should mention that I was able to ssh into the duckiebot once, using quackquack password, but I can't get it to work again.

MMAzeem3, 2021-09-16:
Update: I was able to make it work. Not sure what the problem was exactly but I think my ethernet cable may have been damaged because it worked with a new one

marksoehner, 2021-09-16:
We followed this documentation   https://docs.duckietown.org/daffy/opmanual_duckiebot/out/duckiebot_network.html

the hostname of the duckiebot is duck29

Our ethernet cord also looks a bit damaged, but I don't know if that is what is causing the issue.  I was able to ping the duckiebot, set up the dashboard, and look through the duckiebot camera.

18r441m, 2021-09-16:
I'm closing the issue since it seems both teams resolved the issue

Jennak1, 2021-09-17:
I haven't been able to fix this issue.

18r441m, 2021-09-17:
@Jennak1 im reopening the issue. 

Firstly, check to ensure a physical connection between your computer and the dukciebot. The Ethernet port on the duckiebot has a light indicator, ensure that light is on. 

If the above is true, go ahead and ping the duckiebot. Do so by running: ping <duckiebot-hostname>.local. Let me know if you get any response.

Jennak1, 2021-09-17:
The Ethernet port was flashing. I ran ping duck#.local and didn't get anything. I did not see this specific command in the documentation. I will try this.

18r441m, 2021-09-17:
The command I mentioned is the same one you ran. I forgot to ask but did you run dts fleet discover? and was there a response?

Jennak1, 2021-09-17:
yes, we ran that before trying to bridge ethernet and got nothing. And after I set up the bridge for the ethernet I did it again and nothing. Pretty confident I set up the bridge correctly, but as soon as I tried to do the first ssh command it would not work.

Jennak1, 2021-09-17:
As soon as I ran dts fleet discover just now the status came up to ready but the dashboard still displays down.

brendanreeder, 2021-09-18:
Hey ibrahim, I am also having issues with ssh into the duckie@duck2.local I am  getting the error "kex_exchange_identification: read: Connection reset by peer" I've checked google and they say to change the host.deny and .allow files. however no luck. any suggestions?

18r441m, 2021-09-18:
@brendanreeder restart the duckiebot but unplugging the battery and plugging it back in.

closed 2021-09-20 by 18r441m