18r441m/CSCE274 #13: Ubuntu/SD failure

valerieduffey, 2021-09-16:
My virtual machine will run,  but ubuntu doesn’t open. I have tried closing the virtual machine and restarting my computer, but nothing will help. It comes up as a blank screen.
I believe we messed up the SD installation. When prompted for the size of the SD we gave 32, and nothing was found. However, there was one that said 50 and was the correct name, so we choose it. Then my system crashed, and I haven’t been able to open Ubuntu since. We couldn't see the root from the SD on my computer, and when used the school computer it showed. However, there was nothing on it.

18r441m, 2021-09-16:
It seems like you wiped your vm. You would have to start a new vm and reinstall all the packages including dts.

closed 2021-09-16 by 18r441m

18r441m, 2021-09-16:
Also, a vm is not recommended for this course

valerieduffey, 2021-09-18:
Would using the computers in the lab be a better option than the VM?