18r441m/CSCE274 #21: Ethernet Connection

antoniow2, 2021-09-22:
We made the connection for Ethernet:

[image]
 
However, we are receiving an error saying we cannot resolve host name when trying to do ssh duckie@duck19.local. 

Can you assist?

18r441m, 2021-09-22:
Check the RPI Ethernet port LED indicator, is it on?

antoniow2, 2021-09-22:
Yes Sir, it is on.

18r441m, 2021-09-22:
Run: dts fleet discover

and let me know if the duckiebot is listed?

antoniow2, 2021-09-22:
No Sir, we tried that and it just sits on a screen with nothing listed below.

AlexFischer02, 2021-09-22:
I am having the same issue but dts fleet discover shows my bot as "ready"

18r441m, 2021-09-22:
Check RPI LED indicators and let me know if the green is flashing, which indicates Micro SD activity. Refer to #17

18r441m, 2021-09-22:
@AlexFischer02 stop fleet discover, run it again and share the output

antoniow2, 2021-09-22:
No Sir, it was flashing yellow

18r441m, 2021-09-22:
[image]

in the connection editor, what does the last used say? when you connect the duckiebot to your laptop, that should say "now"

antoniow2, 2021-09-22:
Yes Sir, it was that one and it's another one under it.

[nm-c-e]<https://protect2.fireeye.com/v1/url?k=f72f2f21-a8b417d7-f72f61e0-8627ced94d24-b58d5d5b8ffbf8f7&q=1&e=5d28ed0c-9262-4955-ae24-b5329b3a5231&u=https%3A%2F%2Fuser-images.githubusercontent.com%2F38477627%2F134266418-612b22e2-636f-4db9-9ce5-9ef27411ee16.png>

AlexFischer02, 2021-09-22:
[image]
When i run dts fleet discover this is what I get. I cannot open the dashboard. I have tried http://duck18/ and http://duck18.local/

AlexFischer02, 2021-09-22:
[image]
For further reference this is what I am getting from the dts fleet discover command (top left), all ssh commands (bottom left), and the dashboard page (right).

18r441m, 2021-09-22:
@AlexFischer02 & @antoniow2 Reboot the rebot and try again

AlexFischer02, 2021-09-22:
[image]
Now the ssh is the bottom left is asking for the password. Should I know/have set a password?

AlexFischer02, 2021-09-22:
The dashboard also says "up" but the 502 error is still there

18r441m, 2021-09-22:
@AlexFischer02 the default password is the documentation. It’s “quackquack”

AlexFischer02, 2021-09-22:
The password "quackquack" is returning "Connection closed by 10.42.0.61 port 22"

18r441m, 2021-09-22:
@AlexFischer02 email me and reference this issue.

@antoniow2 did you resolve the issue?

antoniow2, 2021-09-22:
No Sir,

My partner had to leave for the night but after we rebooted it still wouldn't work.

18r441m, 2021-09-22:
@AlexFischer02 and @antoniow2 stop by the lab today during office hours and I'll give you new Micro SD Cards and that should resolve some of the issues

closed 2021-09-22 by 18r441m