18r441m/CSCE274 #22: Operational Keyboard Control Issues

jjadned, 2021-09-22:
We have everything set up, up to keyboard control but when we run the command, it has the download the GUI tools for the controller and has been downloading for over 30 minutes. 

dts duckiebot keyboard_control ROBOT_NAME

jjadned, 2021-09-22:
It's also happening with the camera calibration.

jjadned, 2021-09-22:
and the GUI tools

18r441m, 2021-09-22:
Did you natively boot linux or are your using (vm or wsl)?

jjadned, 2021-09-22:
Yes, we did.

18r441m, 2021-09-22:
ok. Did you bridge the connection to the duckiebot?

jjadned, 2021-09-22:
Yes, we did.

jjadned, 2021-09-22:
We got it to download and now we're getting 30 different issues on getting it to move.

```Open the Portainer interface and check the running containers. You should see one called dt18_03_roscore_duckiebot-interface_1.

You can also determine this by running:

$ docker -H ROBOT_NAME.local ps
``` 

We followed this and it's 2nd step to update it. and none of it is working

jjadned, 2021-09-22:
What it’s doing now, no clue what this is.

jjadned, 2021-09-22:
https://youtube.com/shorts/4JnIo1uTN0Q?feature=share

18r441m, 2021-09-22:
Can you successfully ssh?

jjadned, 2021-09-22:
Did we have to SSH before we could do any kind of motor control?

18r441m, 2021-09-22:
no. but it seems like the dashboard docker container is not healthy. You can rm that container and start a healthy one on the RPI. Also, it seems like your Micro SD Card might be giving you issue (email me and we can go over that)

jjadned, 2021-09-22:
I don’t know what any of that meant expect for the last part. Jack emailed you as we left the lab and I’m CC’d on that if you just want to email that chain.

closed 2021-09-22 by 18r441m