18r441m/CSCE274 #42: Error with dts devel build -H

valerieduffey, 2021-10-05:
[image]

Our Duckie is up and ready, but at step 11/59 it stops and gives us this error. We are in my repo. Should we be in a different directory? Or is there something else we are missing?

valerieduffey, 2021-10-05:
We no longer get the error, however, it freezes after 11/59. We have waited upwards of 10 mins and nothing changed.

valerieduffey, 2021-10-06:
Yes, we were connected to eduroam and our ethernet was connected.

18r441m, 2021-10-06:
Can you join (Teams) office hours now?

valerieduffey, 2021-10-06:
I can't I am in physics right now, and then I got straight to another class.

18r441m, 2021-10-06:
Word. Let's move this to email

closed 2021-10-06 by 18r441m