18r441m/CSCE274 #59: Issue Building Container to Robot: "dts devel build -H duck34.local"

ethannhang, 2021-10-20:
We run this command: **dts devel build -H duck34.local**

We get the following errors, then followed resolutions given to similar issues, which was to rerun it after the error and to wait 15-20 minutes. Even after multiple times of rerunning and waiting a few hours, nothing worked. 

[image]
[image]

Before we began building, we ensured that the duckiebot was connected to the internet and that the dashboard was up. We also noticed that our SD card light would turn off (pictured) while it is building or nothing would happen after an hour.

[image]
.

Jennak1, 2021-10-22:
No. Try unplugging the battery, might work after that

18r441m, 2021-10-31:
Was resolved.

closed 2021-10-31 by 18r441m