18r441m/CSCE274 #65: Duckiebot won't move

KHetzel00, 2021-11-05:
We came in today and found that the we cant use the dts duckiebot keyboard_control program to move our duckiebot, the lane follow demo also will not run, nor can we find its files on our duckiebot.

Strangely enough we tested to see and both of our launch files from project2 do make the duckiebot move.

This happened after we were forced to get an update after trying to use dts fleet discover
It should also be noted that our duckiebot is low on power and we are going to charge it over night and try again tomorrow to see if that fixes anything

Any help would be great

KHetzel00, 2021-11-06:
Even upon re-flashing the SD card the duckiebot still wouldnt move when using the joystick

ZeHolyQofPower, 2021-11-08:
I had a similar issue and solved it this way: Gazdecki's Paranoia List to get keyboard_control working again in DTS 5.2.19
I hope this helps you in some way.

KHetzel00, 2021-11-09:
That did actually help! Running the controls in headless mode has now given me the ability to move it. Howerver another issue has cropped up, I believe due to the same update that seems to have broken everything else. 

As I reflashed my duckiebot in hopes of fixing it, I need to recalibrate the wheels and camera, however the calibration window will only appear with a black screen and the only help given on the documentation is

 "If you only see a window with black screen during the startup, try to resize the window manually using cursor, and you should see the window content correctly."

This however hasnt worked 

If anyone else has had this issue after reflashing and fixed it any help would be great, although I odnt know if i should make this a brand new issue

KHetzel00, 2021-11-09:
Calibration fixed itself

closed 2021-11-09 by KHetzel00