18r441m/CSCE274 #68: Robot will not move

STOMATOPOD, 2021-11-07:
When keyboard control is started with the same command: "dts duckiebot keyboard_control duck4," which was used for past assignments, the process executes smoothly and the virtual joystick appears, but the robot does not respond to key presses. 

The Lane Follow demo does not make the robot move, even when the start key is pressed, which is unsurprising given that the joystick does not appear to work. Core container was started with: "docker -H duck4.local run --rm --name test_lane_follow -v /data:/data --network=host -it duckietown/dt-core:daffy-arm32v7 bash"

Also, both launch files created for project 2 no longer work, even though the files have not been modified. The robot builds and executes the launch files, but does not move for the full duration of the process. Again, no error messages are displayed to the terminal. The command used to launch the container on the robot for project: "dts devel run -H duck4.local -s -M --cmd bash"

ZeHolyQofPower, 2021-11-08:
Hey I had a similar issue. Check out how I solved it and let me know what worked or didn't:
https://docs.google.com/document/d/1nCIKV_8BW_o9nJZm6mtiSCc7Kl44S1-eMkl-3xwmh3M/edit?usp=sharing

18r441m, 2021-11-26:
Rebooting the robot's RPI should fix the problem. We went over that during lecture.

closed 2021-11-26 by 18r441m